MicrosoftDocs/windows-driver-docs-ddi #501: Dead link

wizebin, 2019-05-25:
CreateObjectAttributes
https://technet.microsoft.com/en-us/ff547804%28v=vs.94%29

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dbc76c93-3507-1f66-e586-5b28fed5023a
* Version Independent ID: b2220ab7-09e0-7425-2582-93af10f9e462
* Content: [FltCreateCommunicationPort function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltcreatecommunicationport)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltcreatecommunicationport.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltcreatecommunicationport.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-05-28:
@wizebin , thanks for reporting this. @tedhudek , will reassign to you - this is a result of wudfwdm.h being deleted from WDK.

tedhudek, 2019-06-20:
Thanks @wizebin for the feedback and apologies for the delay getting back to you.  This is now fixed.  I updated all links in the doc repo to point to the new page, which is:

https://docs.microsoft.com/en-us/windows/desktop/api/ntdef/nf-ntdef-initializeobjectattributes

closed 2019-06-20 by tedhudek