MicrosoftDocs/windows-driver-docs-ddi #506: How big should the buffer be?

BrianCatlin, 2019-05-27:
The docs state that the size of the buffer must be at least "sizeof (FILE_FS_ATTRIBUTE_INFORMATION)", but that isn't useful for allocating a buffer. What if I want the file system name, which is tacked on the end? Making two calls is ridiculous. What is the maximum length of a file system name?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c1946dad-51f1-6553-d6ac-ceba66a3a733
* Version Independent ID: 3e40a24e-7072-d101-980d-0b8f034e3c84
* Content: [_FILE_FS_ATTRIBUTE_INFORMATION (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/ns-ntifs-_file_fs_attribute_information)
* Content Source: [wdk-ddi-src/content/ntifs/ns-ntifs-_file_fs_attribute_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ns-ntifs-_file_fs_attribute_information.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-06-08:
@BrianCatlin , I've updated the page to give a better handle on why two calls are necessary in query situations like this. A robust filter will make two calls. If you want to avoid two calls and are working on a filter that is limited to interact with in-box file systems only: currently, the largest in-box file system name is 18 bytes (9 characters; can't guarantee that FSs with longer names won't be introduced in the future).

closed 2019-06-08 by lorihollasch

BrianCatlin, 2019-06-08:
Thanks for the clarification. 
UNICODE_STRINGs use a USHORT for the Length and MaximumLength, so the maximum string size would be 65,536 bytes, or 32,768 characters. I would be greatly surprised if the file systems supported names larger than that