MicrosoftDocs/windows-driver-docs-ddi #508: Note section needs to clarify about version 3

sarojm-MSFT, 2019-05-28:
Returned version number is always 1 for input version of 1\2\3.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7a882f30-8b53-d823-3e89-ebe23f6cd3b5
* Version Independent ID: b3f9df99-32e7-bc99-e05c-3ca3f1b0e09d
* Content: [IoGetDmaAdapter function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-iogetdmaadapter#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-iogetdmaadapter.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-iogetdmaadapter.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-06-28:
Thanks for the feedback @sarojm-MSFT, apologies for the delay.

Are you saying that they set DeviceDescription->Version to the equivalent values for 1, 2, or 3 (DEVICE_DESCRIPTION_VERSION1, etc.)  but **IoGetDmaAdapter** always returns DMA_ADAPTER->Version == 1?

sarojm-MSFT, 2019-06-28:
Yes.

Regards
Saroj Mohapatra

tedhudek, 2019-07-19:
Thanks for the suggestion, @sarojm-MSFT, I've made the change in the doc source.  Give it an hour or two to propagate to the live site.

closed 2019-07-19 by tedhudek

sarojm-MSFT, 2019-07-19:
Nice. Thanks for the help.

Regards
Saroj Mohapatra

Closed #508<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FMicrosoftDocs%2Fwindows-driver-docs-ddi%2Fissues%2F508&data=02%7C01%7CSaroj.Mohapatra%40microsoft.com%7Cce9e73abc5394282a37f08d70c84b9c7%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636991636507253415&sdata=Ozs9QCn5q%2B7hZnotQmcaXzD7IfDTqoROYmakbzYiEl4%3D&reserved=0>.