MicrosoftDocs/windows-driver-docs-ddi #510: Dead Links

SolarGranola, 2019-05-28:
In the Network Driver Headers section, Checksum.h and Checksumtypes.h have dead links.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6b12e0d9-496a-b3e0-313e-d2c951e502e3
* Version Independent ID: 799e2f8e-7af3-c5ed-3d66-bee8eebc9f4c
* Content: [Networking drivers for Windows Vista and later - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/_netvista/#feedback)
* Content Source: [wdk-ddi-src/content/_netvista/index.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/_netvista/index.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @cdwilson
* Microsoft Alias: **windowsdriverdev**

duncanmacmichael, 2019-05-30:
Thanks @SolarGranola, I appreciate the sharp eye. There is a known issue at the moment with getting the content from those headers to render on the site properly, and I've filed an internal bug accordingly. I'll go ahead and close this issue, but thanks for bringing this up.

closed 2019-05-30 by duncanmacmichael

cdwilson, 2019-07-23:
@SolarGranola @EliotSeattle @duncanmacmichael hey guys, I'm getting spammed by a bunch of emails because my GitHub login (@cdwilson) keeps getting referenced in the GitHub login field above.   Is there a way you can fix whatever keeps adding me to these issues?  Here's another one #590

EliotSeattle, 2019-07-23:
Hi Chris (@cdwilson)
I'm really sorry about this. We seem to be conflating your GitHub user name with the email address of our engineer. I'll get someone on this to fix it as soon as we can. I apologize for this and any future mail you get until I get this straightened out.
Eliot