MicrosoftDocs/windows-driver-docs-ddi #517: Code sample is wrong

billti, 2019-06-02:
This doc page is for CreateFunctionAlias, but the code sample shows a call to RegisterFunctionAlias (or which there is no such method on the interface).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6edbaf3a-86e8-e8d2-a099-11a8485e9bd8
* Version Independent ID: e14ca67f-7ab3-e762-34bf-45b4f1b784a0
* Content: [IDebugHostExtensibility::CreateFunctionAlias (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-idebughostextensibility-createfunctionalias)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idebughostextensibility-createfunctionalias.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idebughostextensibility-createfunctionalias.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-11:
@billti - Thanks for this issue, it does look like the sample is not right. I'm investigating with dev team what the fix should be.

DOMARS, 2020-10-29:
@billti - Sorry for the delay here. I updated the sample code to use CreateFunctionAlias, as RegisterFunctionAlias does not seem to exist. Thanks for the sharp eye, and for this issue.

closed 2020-10-29 by DOMARS