MicrosoftDocs/windows-driver-docs-ddi #519: There should be more information about the ModelObjectKind enumeration

jazzdelightsme, 2019-06-04:
a link to a page about it, and possibly even an "inline" description

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3cf566d8-3d88-d798-fb1c-26ed5b6234ec
* Version Independent ID: 634e6f3f-5b9f-0a46-7f74-520650748a01
* Content: [IModelObject::GetKind (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-imodelobject-getkind#feedback)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-imodelobject-getkind.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-imodelobject-getkind.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

jazzdelightsme, 2019-06-04:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/ne-dbgmodel-modelobjectkind

DOMARS, 2019-06-10:
@jazzdelightsme - Thanks for the review, I have added the link. I will close this issue when the changes are live.

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS