MicrosoftDocs/windows-driver-docs-ddi #520: There should be a link to info about the Location structure

jazzdelightsme, 2019-06-04:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/ns-dbgmodel-location

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5f3de874-7096-3d08-14ff-5252357f9df8
* Version Independent ID: 2c9f70f5-68b0-0dba-785c-c4b1860feede
* Content: [IModelObject::GetLocation (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-imodelobject-getlocation#feedback)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-imodelobject-getlocation.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-imodelobject-getlocation.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-10:
@jazzdelightsme - Thanks for the review, I have added the link. I will close this issue when the changes are live. You are always welcome to send a PR for changes like this.

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS