MicrosoftDocs/windows-driver-docs-ddi #521: There should be a link on this page to IDebugHostType

jazzdelightsme, 2019-06-04:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nn-dbgmodel-idebughosttype

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4a67a207-a4f7-8de4-755d-feb9fd74489d
* Version Independent ID: 2af2e42a-dae0-a0cc-2b07-58f4b2594f55
* Content: [IModelObject::GetTypeInfo (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-imodelobject-gettypeinfo#feedback)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-imodelobject-gettypeinfo.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-imodelobject-gettypeinfo.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-10:
@jazzdelightsme - Thanks for the review, I have added the link. I will close this issue when the changes are live. You are always welcome to send a PR for changes like this.

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS