MicrosoftDocs/windows-driver-docs-ddi #522: There should be a link to the SymbolKind enumeration

jazzdelightsme, 2019-06-04:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/ne-dbgmodel-symbolkind

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 623e2abf-9661-adf1-de74-b79c601bd8b6
* Version Independent ID: 561e32e4-0077-f4ff-ad0a-fdc45f48c822
* Content: [IDebugHostType::GetSymbolKind (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-idebughosttype-getsymbolkind)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idebughosttype-getsymbolkind.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idebughosttype-getsymbolkind.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-10:
@jazzdelightsme - Thanks for the review, I have added the link. I will close this issue when the changes are live. You are always welcome to send a PR for changes like this.

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS