MicrosoftDocs/windows-driver-docs-ddi #523: There should be a link to the TypeKind enumeration

jazzdelightsme, 2019-06-04:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/ne-dbgmodel-typekind

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 8ced7742-c8b8-1ce6-41aa-1288587488b0
* Version Independent ID: bb420124-ef58-0a85-7679-21b50281cd46
* Content: [IDebugHostType::GetTypeKind (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-idebughosttype-gettypekind)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idebughosttype-gettypekind.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idebughosttype-gettypekind.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-10:
@jazzdelightsme - Thanks for the review, I have added the link. I will close this issue when the changes are live. You are always welcome to send a PR for changes like this.

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS