MicrosoftDocs/windows-driver-docs-ddi #524: missing description for the TypeExtendedArray enumerand

jazzdelightsme, 2019-06-04:
title says it all

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 86ed8ede-6367-2fff-6712-7a6652db0725
* Version Independent ID: dfdea968-6ccb-4a9e-d446-99b0df0e27c0
* Content: [TypeKind (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/ne-dbgmodel-typekind)
* Content Source: [wdk-ddi-src/content/dbgmodel/ne-dbgmodel-typekind.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/ne-dbgmodel-typekind.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-10:
@jazzdelightsme - I have added this description from the header for the  TypeExtendedArray field.

The type is an array which cannot be expressed as TypeArray. This is due to things such as dynamic sizes, dynamic bounds, etc...  CLI arrays are represented as TypeExtendedArray.

I will close this issue when the changes are live. You are always welcome to send a PR for changes like this.

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS