MicrosoftDocs/windows-driver-docs-ddi #526: Needs elaboration: "there are certain names that have defined semantic values"

jazzdelightsme, 2019-06-04:
What are those names, and what are the corresponding values?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bc9e5fb8-5b5f-cfe0-2727-cbbd5a5f59a3
* Version Independent ID: ec8d81cd-5a86-582d-e5a9-aed1207692cd
* Content: [IKeyStore (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nn-dbgmodel-ikeystore)
* Content Source: [wdk-ddi-src/content/dbgmodel/nn-dbgmodel-ikeystore.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nn-dbgmodel-ikeystore.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-10:
@aluhrs13 - Andy can you help with this request? We make a pretty broad statement about existing key pairs, and it would be good to provide a few examples or otherwise clarify what the constraints are or what the existing key pairs would be.

By convention only, the values in a metadata defined key store are restricted to basic values (intrinsics and property accessors) *While a key store can have an arbitrary number (and arbitrary naming) of keys, there are certain names that have defined semantic values.*

Does this imply that a developer should not duplicate the existing pairs? If so, can we provide a dx command or code snip to introspect the objects to learn these, or provide a similar process to list them?

DOMARS, 2019-06-10:
@jazzdelightsme - I will add *this* text to provide additional information about the defined semantic values:

While a key store can have an arbitrary number (and arbitrary naming) of keys, there are certain names that have defined semantic values. *For a list of those values, see [Debugger Data Model Metadata Interfaces](https://docs.microsoft.com/windows-hardware/drivers/debugger/data-model-cpp-additional-interfaces#-debugger-data-model-metadata-interfaces).*

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS