MicrosoftDocs/windows-driver-docs-ddi #527: There should be a link to the IDataModelConcept page

jazzdelightsme, 2019-06-04:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nn-dbgmodel-idatamodelconcept

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0adff9eb-1553-a20a-6313-e8ed6482dc84
* Version Independent ID: e08853ba-e811-a758-d546-6b653f9a83c2
* Content: [IDataModelManager::CreateDataModelObject (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-idatamodelmanager-createdatamodelobject)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idatamodelmanager-createdatamodelobject.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idatamodelmanager-createdatamodelobject.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-10:
@jazzdelightsme - I have added the link here and also to the IDataModelManager2 version. I will close this issue when the changes are live. You are always welcome to send a PR for changes like this.

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS