MicrosoftDocs/windows-driver-docs-ddi #528: Typo in the description of "context" parameter: missing "will"

jazzdelightsme, 2019-06-04:
"The debug host context which be associated with the newly created synthetic object."

Should probably be something like:

"The debug host context which **will** be associated with the newly created synthetic object."

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 60c4c1ae-7f91-9fbf-8b24-2be529ed8618
* Version Independent ID: 13ac00e4-6b6b-bd48-acc1-7d625b64cd7c
* Content: [IDataModelManager::CreateSyntheticObject (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-idatamodelmanager-createsyntheticobject)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idatamodelmanager-createsyntheticobject.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idatamodelmanager-createsyntheticobject.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-10:
@jazzdelightsme - That wording is fixed here and in IDataModelManager2.  I will close this issue when the changes are live. You are always welcome to send a PR for changes like this.

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS