MicrosoftDocs/windows-driver-docs-ddi #529: This page should compare/contrast with the IDataModelManager interface

jazzdelightsme, 2019-06-04:
I.e. which methods are new? Which behave differently?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f721db8d-4f4e-c13e-a451-15099bf4d7d4
* Version Independent ID: 3b034261-2a28-be61-8522-c0e9e093695e
* Content: [IDataModelManager2 (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nn-dbgmodel-idatamodelmanager2#feedback)
* Content Source: [wdk-ddi-src/content/dbgmodel/nn-dbgmodel-idatamodelmanager2.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nn-dbgmodel-idatamodelmanager2.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-11:
@jazzdelightsme - I'm working with the dev team to see if we can describe the interface differences.

DOMARS, 2019-06-12:
@jazzdelightsme 
By design, the follow is true of all version 2 interfaces. 

*This version 2 of the interface supports all of the previous methods with identical signatures and includes additional new methods that provide added functionality. The new methods are listed in the header at the end of the section for that interface.*

I added this text to all of the version 2 interfaces. Thanks for spending the time to create the issues for the dbgmodel.

closed 2019-06-12 by DOMARS