MicrosoftDocs/windows-driver-docs-ddi #530: Need a link to the IDataModelNameBinder topic

jazzdelightsme, 2019-06-04:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nn-dbgmodel-idatamodelnamebinder

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a9080d10-05b2-ad02-0fdd-55ff87af1136
* Version Independent ID: 7b2fa378-035a-04a8-1b6e-ee9fdaa7e792
* Content: [IDataModelScriptManager::GetDefaultNameBinder (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-idatamodelscriptmanager-getdefaultnamebinder)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idatamodelscriptmanager-getdefaultnamebinder.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idatamodelscriptmanager-getdefaultnamebinder.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-10:
@jazzdelightsme - I have added the link. I will close this issue when the changes are live. You are always welcome to send a PR for changes like this.

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS