MicrosoftDocs/windows-driver-docs-ddi #531: Need a link to the IDataModelScriptProvider topic

jazzdelightsme, 2019-06-04:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nn-dbgmodel-idatamodelscriptprovider

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6249d1b7-8bfa-991a-0771-1ff58dc4a297
* Version Independent ID: e8989fb0-15f7-09ff-5dc6-b414a881fbd9
* Content: [IDataModelScriptManager::RegisterScriptProvider (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-idatamodelscriptmanager-registerscriptprovider)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idatamodelscriptmanager-registerscriptprovider.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idatamodelscriptmanager-registerscriptprovider.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-10:
@jazzdelightsme - I have added the link. I will close this issue when the changes are live. You are always welcome to send a PR for changes like this.

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS