MicrosoftDocs/windows-driver-docs-ddi #533: Need a link to SymbolKind enumeration

jazzdelightsme, 2019-06-05:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/ne-dbgmodel-symbolkind

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e12e300b-e5be-106c-031b-6130d22fad42
* Version Independent ID: d5fc4c83-ed17-6dde-0729-1d46e1ea4da1
* Content: [IDebugHostSymbol::GetSymbolKind (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-idebughostsymbol-getsymbolkind)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idebughostsymbol-getsymbolkind.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idebughostsymbol-getsymbolkind.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-10:
@jazzdelightsme - I have added the link. I will close this issue when the changes are live. You are always welcome to send a PR for changes like this.

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS