MicrosoftDocs/windows-driver-docs-ddi #534: Need to explain difference between this interface and IDebugHostSymbol

jazzdelightsme, 2019-06-05:
title says it all

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c1a18160-41a6-4cf8-c24a-7d57d7ec490d
* Version Independent ID: 1a7008c8-5fd8-f923-7c8d-11d20ec87280
* Content: [IDebugHostSymbol2 (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nn-dbgmodel-idebughostsymbol2)
* Content Source: [wdk-ddi-src/content/dbgmodel/nn-dbgmodel-idebughostsymbol2.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nn-dbgmodel-idebughostsymbol2.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-11:
@jazzdelightsme - I'm working with the dev team to see if we can describe the interface differences.

DOMARS, 2019-06-12:
@jazzdelightsme - By design, the following is true of all version 2 interfaces. 

*This version 2 of the interface supports all of the previous methods with identical signatures and includes additional new methods that provide added functionality. The new methods are listed in the header at the end of the section for that interface.*

I added this text to all of the version 2 interfaces. Thanks for spending the time to create the issues for the dbgmodel.

closed 2019-06-12 by DOMARS