MicrosoftDocs/windows-driver-docs-ddi #535: Need a link to the SymbolSearchInfo structure

jazzdelightsme, 2019-06-05:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/ns-dbgmodel-symbolsearchinfo

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 45f03276-f6b1-56bb-4ba1-7f508d25aebc
* Version Independent ID: 81423d35-8532-b0a0-ab35-b7a7a9e99705
* Content: [IDebugHostSymbol2::EnumerateChildrenEx (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-idebughostsymbol2-enumeratechildrenex)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idebughostsymbol2-enumeratechildrenex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idebughostsymbol2-enumeratechildrenex.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-11:
@jazzdelightsme - I have added the link. I will close this issue when the changes are live. You are always welcome to send a PR for changes like this.

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS