MicrosoftDocs/windows-driver-docs-ddi #536: Need a link to the IDataModelScriptHostContext page

jazzdelightsme, 2019-06-05:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nn-dbgmodel-idatamodelscripthostcontext

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: fdd7b6a1-28bd-dfd6-3bb1-f5a4ef441427
* Version Independent ID: 823550d0-6791-c4f5-4a1e-26613223b38c
* Content: [IDebugHostScriptHost::CreateContext (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-idebughostscripthost-createcontext)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idebughostscripthost-createcontext.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-idebughostscripthost-createcontext.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-11:
@jazzdelightsme - I have added the link. I will close this issue when the changes are live. You are always welcome to send a PR for changes like this.

DOMARS, 2019-06-11:
@jazzdelightsme - The topic updates are live, so closing this issue. Thanks for the issue and your interest in the DbgModel docs.

closed 2019-06-11 by DOMARS