MicrosoftDocs/windows-driver-docs-ddi #540: Not present in offline help viewer

PisarevSergey, 2019-06-09:
Not present in offline help viewer

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ebff2f40-62ed-8c2a-58c7-f712146a4e2e
* Version Independent ID: 60ddd91f-76f1-37d1-53a0-8d7e516d08fe
* Content: [FltQueryInformationByName function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltqueryinformationbyname#feedback)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltqueryinformationbyname.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltqueryinformationbyname.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-06-27:
@PisarevSergey , are you using an old CHM to reference WDK documentation? We only publish documentation online through docs.microsoft.com now - we are not generating CHMs anymore. I'm going to close this issue since the refpage is published on the website. Let me know if you are referring to something else, thanks.

PisarevSergey, 2019-06-27:
I am using Help Viewer that comes with Visual Studio 2019

lorihollasch, 2019-06-27:
Thanks for clarifying! I'll reopen this issue and investigate.

lorihollasch, 2019-09-09:
@PisarevSergey, we just published updated offline books. Go ahead and update your HelpViewer docs and all should be good now!

closed 2019-09-09 by lorihollasch