MicrosoftDocs/windows-driver-docs-ddi #541: Not helpful

BrianCatlin, 2019-06-09:
WHY would a driver create this routine? Is there an advantage to creating a context allocation routine this way versus just a simple routine to perform the context allocation and initialization?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c387aa9f-b1d8-145c-e1ef-cf084af44664
* Version Independent ID: 66f761e6-bbb9-596a-8d4f-f81eb5d8b84b
* Content: [PFLT_CONTEXT_ALLOCATE_CALLBACK (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nc-fltkernel-pflt_context_allocate_callback)
* Content Source: [wdk-ddi-src/content/fltkernel/nc-fltkernel-pflt_context_allocate_callback.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nc-fltkernel-pflt_context_allocate_callback.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-06-27:
@briancatlin, this page states "For the rare cases that a minifilter driver must perform its own context allocation, it can specify a routine of type PFLT_CONTEXT_ALLOCATE_CALLBACK as the ContextAllocateCallback routine ...". If your driver doesn't need to do its own context allocation, then absolutely use the simple FltXxxContext route.

closed 2019-06-27 by lorihollasch

BrianCatlin, 2019-06-27:
That still doesn't answer the question of why a driver would need to do its own context allocation. A few examples would help