MicrosoftDocs/windows-driver-docs-ddi #544: Auto Markup strikes again.

rodwiddowson, 2019-06-11:
I cannot do a PR for this because the bug is in the auto-generated bit of the markup so...
* The parameter shouldn't be called FL (although of course that's what it is called in `ntifs.h` so we may be SOL)
* The function is is pretending to be is of return type `BOOLEAN`
* The return value is a `BOOLEAN` which indicates whether any byte range locks exist for the specified file.

Extra marks for crossreferencing `FsRtlAreThereCurrentOrInProgressFileLocks` (https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-_fsrtl_advanced_fcb_header-fsrtlaretherecurrentorinprogressfilelocks which is another auto-generated markup disaster, but thats for another day.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e62b54d2-40ea-33b8-0842-f7c8a9f34da4
* Version Independent ID: 6b94eaba-0ffc-de0a-ad0e-4e4b5e547346
* Content: [FsRtlAreThereCurrentFileLocks macro (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-fsrtlaretherecurrentfilelocks)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-fsrtlaretherecurrentfilelocks.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-fsrtlaretherecurrentfilelocks.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

rodwiddowson, 2019-06-11:
Actually I see now that all the macros which return anything are similarly wrong eg `FsRtlIsAnsiCharacterWild` s this is probably known about

lorihollasch, 2019-06-28:
@rodwiddowson:
- You're right that we are SOL on parameter name since the syntax block is auto-generated based on the header file.
- I added the return values in Remarks (the "void" definition requires that Return Value be none).
-I have entered an internal task to check all FsRtl macros and add return value information as required. (Feel free to whittle away at them through PRs between builds, etc if you're so inclined. :)
Thanks for reporting!

closed 2019-06-28 by lorihollasch