MicrosoftDocs/windows-driver-docs-ddi #547: Needs links to other pages

englishb, 2019-06-12:
There is not enough context information here.  What APIs is DXGK_MULTIPLANE_OVERLAY_VSYNC_INFO2 used with?  Does the driver receive this structure from a DxgkDdi function?  Does the driver pass this to the OS in a pfn*Cb function?  Which ones, and please link to the pages for those functions.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 10db98ee-7218-eb19-a427-e0d12bb4a450
* Version Independent ID: c28d15a9-c873-2978-a3fe-4905f995f3c6
* Content: [_DXGK_MULTIPLANE_OVERLAY_VSYNC_INFO2 (d3dkmddi.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/d3dkmddi/ns-d3dkmddi-_dxgk_multiplane_overlay_vsync_info2)
* Content Source: [wdk-ddi-src/content/d3dkmddi/ns-d3dkmddi-_dxgk_multiplane_overlay_vsync_info2.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3dkmddi/ns-d3dkmddi-_dxgk_multiplane_overlay_vsync_info2.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

englishb, 2019-06-12:
The reference to DXGKDDI_POSTMULTIPLANEOVERLAYPRESENT on this page should also be a hyperlink.  I keep having to go back to google to search for the next MSDN page.

EliotSeattle, 2019-06-12:
Thanks for letting us know, @englishb. I've added the link. Should be live in a few moments.

englishb, 2019-06-12:
Okay, thank you for adding the link.  However, this does not address the original issue I opened, only the comment I made afterward.

EliotSeattle, 2019-06-12:
Oops. My bad, @englishb 
I saw the notification about the link in my inbox and jumped on it, trying to keep the low hanging fruit off of Lori's plate. I'll leave the original request for more info in her capable hands, and have reopened this Issue.
Cheers,

Eliot

englishb, 2019-06-12:
Thanks Eliot.

lorihollasch, 2020-05-29:
@englishb, I've added technical content to this page and to DXGKARGCB_NOTIFY_INTERRUPT_DATA, which this structure is a member of. Thanks for reporting the hole.

closed 2020-05-29 by lorihollasch