MicrosoftDocs/windows-driver-docs-ddi #548: Some file systems don't have a GUID name

BrianCatlin, 2019-06-13:
Raw and network file systems don't have a volume GUID name. It would be nice if there were a table that listed which inbox file systems don't have a volume GUID name

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d1e71c47-8235-65d8-a247-6704e00e10be
* Version Independent ID: a2c1a764-676f-3c8e-ec71-20c164758030
* Content: [FltGetVolumeGuidName function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltgetvolumeguidname)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetvolumeguidname.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetvolumeguidname.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

rodwiddowson, 2019-06-13:
Worth pointing out that show volumes which IIRC are definitely NTFS do not have GUIDs. So it’s nothing to do with the file system...

lorihollasch, 2019-07-12:
@briancatlin, @rodwiddowson is correct: volume GUIDs are file system-independent. Check out the Remarks section for more information, thanks!

closed 2019-07-12 by lorihollasch