MicrosoftDocs/windows-driver-docs-ddi #552: Add note about if this is sent to a controller instead of target

csm10495, 2019-06-19:
Add note about Path/Target/Lun being all 0xFF in event of IOCTL_SCSI_GET_ADDRESS being sent to a \\.\SCSI#: controller starting on RS5 (Port is filled  out properly). Previous to RS5 it seemed to just fail the DeviceIoControl() call. Document out this behavior change.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 74a811d7-86aa-dd3f-bc40-eb11ed38343b
* Version Independent ID: ebb75d79-5847-89f5-20dd-40e87ccb702f
* Content: [IOCTL_SCSI_GET_ADDRESS (ntddscsi.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddscsi/ni-ntddscsi-ioctl_scsi_get_address)
* Content Source: [wdk-ddi-src/content/ntddscsi/ni-ntddscsi-ioctl_scsi_get_address.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddscsi/ni-ntddscsi-ioctl_scsi_get_address.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-11-05:
@csm10495 , thanks for submitting this feedback. The storage team did indeed expand this IOCTL's support from just the storage disk object to both the storage disk and adapter objects. They have indicated that they'd prefer the page to be left as is, with the expanded support implied.

closed 2019-11-05 by lorihollasch

csm10495, 2019-11-06:
@lorihollasch, Expanded support implied? That seems a bit counter productive to having explicit documentation. Any particular reason why the team would rather not document the behavior change?

lorihollasch, 2019-11-08:
@csm10495 , I'm glad you pushed back on this. I re-read the mail thread with the dev lead, and believe I misinterpreted his feedback. I've updated the page accordingly, thanks!

csm10495, 2019-11-09:
Yay thanks for the clarification on the page!