MicrosoftDocs/windows-driver-docs-ddi #554: Format Rendering Issue

474172261, 2019-06-20:
The format of Members doesn't render well.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e4fa42b5-1632-42e6-8d7d-05778b6eeaa3
* Version Independent ID: ad17836c-e33f-a306-6912-e5ad12d46c6a
* Content: [_FSRTL_ADVANCED_FCB_HEADER (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/ns-ntifs-_fsrtl_advanced_fcb_header#feedback)
* Content Source: [wdk-ddi-src/content/ntifs/ns-ntifs-_fsrtl_advanced_fcb_header.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ns-ntifs-_fsrtl_advanced_fcb_header.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-06-21:
@474172261 , thanks for reporting this. It's a known issue and is due to a bug in our internal documentation build tools. I'm going to close this issue because we're tracking the issue through an internal bug. So sorry for the poor reading experience - you'll have to consult the header file for now!

closed 2019-06-21 by lorihollasch

lorihollasch, 2019-07-29:
@474172261, FYI that our team has fixed the issue with rendering this page. Thanks again for reporting!