MicrosoftDocs/windows-driver-docs-ddi #555: Missing Create Options

rodwiddowson, 2019-06-20:
@lorihollasch 

I'd tried to do this as a PR and gave up since it seems to me that what is in github is not the real source since these words look duplicated across the `FltCreateFileXXX` and probably across `IoCreateFileXXX` and NtCreateFile and `ZwCreateFile`.  They certainly need to be kept in synch.

There are a few CreateOptionsFLags missing.  The one I was looking to was

`FILE_OPEN_NO_RECALL`  - Instruct any offline storage filters *NOT* to provoke a recall as a result of this open.

But I think there are others 
	`FILE_NO_COMPRESSION` (good luck with that)
	`FILE_OPEN_REMOTE_INSTANCE` (this too)
	`FILE_DISALLOW_EXCLUSIVE`
	`FILE_OPEN_FOR_FREE_SPACE_QUERY`

If I was a dev, I would want to say "reserved to Microsoft", but unfortunately that ship has sailed and did so when the bits got into wdm.h

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ab2c39da-36d8-c2c3-c978-eae7733de3c1
* Version Independent ID: db182921-736b-9a62-1af3-79d2b5fb392c
* Content: [FltCreateFileEx function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltcreatefileex)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltcreatefileex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltcreatefileex.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-08-02:
@rodwiddowson , thanks for reporting the missing flags. I was finally able to wrangle descriptions for the missing ones. I've added them to the FltCreateFileXxx pages for now. I've got an internal task filed to update all related pages (I'm gonna rework them such that the duplicate information ends up on just one of the pages, and add links to the other pages). (Hope you're having a great summer!)

closed 2019-08-02 by lorihollasch

rodwiddowson, 2019-08-02:
@lorihollasch thanks for that. The FILE_NO_COMPRESSION option will be very handy,,