MicrosoftDocs/windows-driver-docs-ddi #556: Whatever happened to FsRtlChangeBackingFileObject?

rodwiddowson, 2019-06-20:
@lorihollasch
I went to ask to make a change on this which I've since discovered isn't needed and its not in github?  F1 in devstudio points me to https://msdn.microsoft.com/query/dev15.query?appId=Dev15IDEF1&l=EN-US&k=k(FsRtlChangeBackingFileObject);k(DevLang-C%2B%2B);k(TargetOS-Windows)&rd=true !

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0f39a405-8a1a-8d58-a796-05691153a17c
* Version Independent ID: bf989985-3b5f-0304-f20e-bc1a7a2838ae
* Content: [ntifs.h header - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/)
* Content Source: [wdk-ddi-src/content/ntifs/index.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/index.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

rodwiddowson, 2019-06-20:
also `CcGetFileObjectFromSectionPtrsRef`` (which is the one I did want to suggest a change for)

lorihollasch, 2019-07-31:
Hi @rodwiddowson , our docs build team has fixed some issues and these functions' refpages (and many others) are now back!

closed 2019-07-31 by lorihollasch