MicrosoftDocs/windows-driver-docs-ddi #566: Various invalid urls or .md links that don't exist

csm10495, 2019-06-28:
I wrote a heuristic script that can be run to check this repo for 'invalid links'. Invalid links are http/s links that 404, have ssl issues, or take longer than 60 seconds to load. Invalid md links are to md files that don't appear to be in the repo.

The script outputs two files: invalid_links_in_files.txt and invalid_links_per_file.txt.

The first of which prints files then invalid links within said file.
The second of which prints an invalid link and then the files that have references to it.

Attaching the script and examples of the two output files.

Can we have something like this in a CI process to ensure that docs don't have broken links and references?
[invalid_links_in_files.txt](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/files/3337769/invalid_links_in_files.txt)
[invalid_links_per_file.txt](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/files/3337770/invalid_links_per_file.txt)

```
'''
validate_links.py - Charles Machalow (MIT License)
'''

import re
import collections
import os
import multiprocessing.pool
import multiprocessing

import requests
import traceback

USE_PROCESSES = True
if USE_PROCESSES:
    Pool = multiprocessing.Pool
else:
    Pool = multiprocessing.pool.ThreadPool

DOC_SRC = os.path.abspath(os.path.join(os.path.dirname(__file__), 'wdk-ddi-src'))
THIS_DIR = os.path.abspath(os.path.join(DOC_SRC, os.pardir))

URL_FINDER_REGEX = re.compile(rb'(http[s|].+?)[\s|\"|\)|\(|\|\<)]')
MD_FINDER_REGEX = re.compile(rb'\((.+?\.md)\)')

def getAllFileNames():
    allFileNames = set()
    for root, dirs, files in os.walk(DOC_SRC):
        for name in files:
            allFileNames.add(name.lower())

    return allFileNames

ALL_FILE_NAMES = getAllFileNames()

def linkValid(url):
    url = url.lower()
    if url.endswith('.md'):
        return url in ALL_FILE_NAMES or url.replace('_', '') in ALL_FILE_NAMES
    elif url.startswith('http'):
        try:
            return requests.get(url, timeout=60).status_code != 404
        except:
            print ("Invalid url via exception (%s)" % url)
            traceback.print_exc()
            return False
    else:
        return False

if __name__ == '__main__':
    pool = Pool(128)
    urlsToResults = {}
    filesToUrls = collections.defaultdict(set)
    for root, dirs, files in os.walk(DOC_SRC):
        for name in files:
            fullPath = os.path.join(root, name)
            with open(fullPath, 'rb') as f:
                txt = f.read()
                matches = re.findall(URL_FINDER_REGEX, txt)
                matches.extend(re.findall(MD_FINDER_REGEX, txt))
                for url in matches:
                    if b' ' not in url:
                        url = url.decode()

                        if url.lower().endswith('.md'):
                            # hack. (todo: fix regex)
                            url = url.split('(')[-1]
                            url = url.split('/')[-1].split('\\')[-1]

                        filesToUrls[os.path.relpath(fullPath, DOC_SRC)].add(url)
                        if url not in urlsToResults:
                            print ("Testing url: %s" % url)
                            #if url.endswith('.md'):
                            #    linkValid(url)
                            urlsToResults[url] = pool.apply_async(linkValid, (url,))

    for idx, (key, value) in enumerate(urlsToResults.items()):
        print ("On %d / %d." % (idx + 1, len(urlsToResults)))
        while True:
            try:
                urlsToResults[key] = value.get(2)
            except multiprocessing.TimeoutError:
                print ("Still stuck on %s" % key)
                continue
            else:
                break

    pool.close()
    pool.join()

    with open(os.path.join(THIS_DIR, 'invalid_links_in_files.txt'), 'w') as f:
        f.write ("The following files have the following invalid links:\n")
        for filePath, urls in filesToUrls.items():
            if urls:
                printedHead = False
                for i in set(urls):
                    if not urlsToResults[i]:
                        if not printedHead:
                            f.write (filePath + '\n')
                            printedHead = True
                        f.write ('  ' + i + '\n')

    with open(os.path.join(THIS_DIR, 'invalid_links_per_file.txt'), 'w') as f:
        f.write ("The following invalid links are found in the following files\n")
        for url, result in urlsToResults.items():
            if not result:
                f.write(url + '\n')
                for filePath, urls in filesToUrls.items():
                    if url in urls:
                        f.write ('  ' + filePath + '\n')

    print ("Done")

```

tedhudek, 2019-10-08:
Thanks @csm10495 for the great feedback.  We're figuring out how best to address it with limited resources.

tedhudek, 2019-10-10:
Hi again @csm10495, I created an internal work item on myself to work on broken links in our docs.  Really appreciate your taking the time to write and run your script on this content set.

closed 2019-10-10 by tedhudek