MicrosoftDocs/windows-driver-docs-ddi #568: how to use those scripts can someone tell me pls

cggiy, 2019-06-30:
how to use those scripts can someone tell me pls

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5b0d7d66-2edf-2b17-d671-c4003c593324
* Version Independent ID: 63970693-25e5-f437-c2e6-71c5f24ed16b
* Content: [NdisOffloadTcpSend function (ndischimney.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ndischimney/nf-ndischimney-ndisoffloadtcpsend#feedback)
* Content Source: [wdk-ddi-src/content/ndischimney/nf-ndischimney-ndisoffloadtcpsend.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ndischimney/nf-ndischimney-ndisoffloadtcpsend.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-07-01:
Hi @cggiy, I'm sorry but I'm not sure what you mean here. Can you be more specific? What script are you talking about? Also, just to note it, the TCP/IP Chimney Offload feature is deprecated and should no longer be used in your miniport driver.

cggiy, 2019-07-01:
HI Thanks you very much cuz of reply. i mean like
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ndischimney/ns-ndischimney-_ndis_tcp_connection_offload_parameters
look
at thee "DelayedAckFrequency" script can be writen in TCPIP in registtry
but like those are C++ scripts and how to use them... should i write all
inj .ini file and drop in C/WIndows/ at there ? or what to do. im trying to
optimize internet connection and TCP packets for sending and receiving so i
need those but i donty understand how ot use them

duncanmacmichael, 2019-07-05:
I see @cggiy, but I think you and I might not be on the same page here. You say:

"im trying to
optimize internet connection and TCP packets for sending and receiving so i
need those but i donty understand how ot use them" 

So that seems to be the problem you're trying to solve. Are you a driver developer? These docs are not for "scripts." They are programming interfaces for developing device drivers. Also, the NDIS Chimney offload is a deprecated feature and should not be used, as Windows does not support it anymore.

If you're trying to optimize your Internet connection, I'd probably recommend checking out IT Pro-oriented information or network management websites, as these doc pages are intended for an entirely different audience.

cggiy, 2019-07-06:
oh ok thanks you sir.

duncanmacmichael, 2019-07-08:
Sure thing. Best of luck with your issue.

closed 2019-07-08 by duncanmacmichael