MicrosoftDocs/windows-driver-docs-ddi #572: The definition is incorrect.

stephenscalzo, 2019-07-07:
The enumeration type definition on this page for _OBJECT_INFORMATION_CLASS is not correct. The ObjectTypeInformation field should have a value of 2 instead of 1.

The fields in order should be:
1. ObjectBasicInformation (value 0).
2. ObjectNameInformation (value 1).
3. ObjectTypeInformation (value 2).
4. ObjectTypesInformation (value 3).
5. ObjectHandleFlagInformation (value 4).
6. ObjectSessionInformation (value 5).
7. ObjectSessionObjectInformation (value 6).
8. MaxObjectInfoClass (value 7).

The enumeration type definition for this document should be replaced with the following:

```
typedef enum _OBJECT_INFORMATION_CLASS {
  ObjectBasicInformation = 0,
  ObjectNameInformation,
  ObjectTypeInformation,
  ObjectTypesInformation,
  ObjectHandleFlagInformation,
  ObjectSessionInformation,
  ObjectSessionObjectInformation,
  MaxObjectInfoClass
} OBJECT_INFORMATION_CLASS;
```

However, if you'd like to keep most of the fields redacted from the documentation, then it should be replaced with the following:

```
typedef enum _OBJECT_INFORMATION_CLASS {
  ObjectBasicInformation = 0,
  ObjectTypeInformation = 2
} OBJECT_INFORMATION_CLASS;
```

For anyone who is wondering how I acquired this information, the enumeration type information for the*_OBJECT_INFORMATION_CLASS is available in the Windows kernel debugging symbols (ntkrnlmp.pdb).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 78703b1a-ae2b-4a2a-f196-e9c62e8fdc55
* Version Independent ID: cf5f65ca-39ad-09ee-96a2-b07a96db8e50
* Content: [_OBJECT_INFORMATION_CLASS (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/ne-ntifs-_object_information_class#feedback)
* Content Source: [wdk-ddi-src/content/ntifs/ne-ntifs-_object_information_class.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ne-ntifs-_object_information_class.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-07-08:
@stephenscalzo , thanks for submitting this. The public _ntifs.h_ header file only lists the two enum values that are currently documented, and we document only what is public.
Our documentation build tool generates the Syntax header, so I cannot add the constants' actual numeric definitions. Devs should stick to the defined enum text-based values and not use the numeric values.

closed 2019-07-08 by lorihollasch

stephenscalzo, 2019-07-08:
Non problemo and thank you for looking into it.

lorihollasch, 2019-07-09:
Reclosing this issue since it appears that my close yesterday didn't take. Comment from yesterday: "@stephenscalzo , thanks for submitting this. The public ntifs.h header file only lists the two enum values that are currently documented, and we document only what is public.
Our documentation build tool generates the Syntax header, so I cannot add the constants' actual numeric definitions. Devs should stick to the defined enum text-based values and not use the numeric values."