MicrosoftDocs/windows-driver-docs-ddi #573: Which level may be called PsGetCurrentThreadTeb?

Grabelkovsky, 2019-07-08:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6a53b79f-a473-9094-51d4-21281b631ff8
* Version Independent ID: 8ffb80f1-0341-5b84-9682-bc8733881ef3
* Content: [PsGetCurrentThreadTeb function (ntddk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/nf-ntddk-psgetcurrentthreadteb)
* Content Source: [wdk-ddi-src/content/ntddk/nf-ntddk-psgetcurrentthreadteb.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nf-ntddk-psgetcurrentthreadteb.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

stephenscalzo, 2019-07-08:
The *PsGetCurrentThreadTeb* routine works by returning a pointer address stored in the *_KTHREAD* structure. The *_KTHREAD* structure that is used is found in the *_ETHREAD* structure for the current thread object (first field).

As far as I am aware, you should be able to use the kernel API whilst running at any interrupt request level, the same way you can use the PsGetCurrentThreadId kernel API whilst running at any interrupt request level.

I would suggest waiting for an official Microsoft employee to confirm this before blindly relying on what I've said. It's always better to be safe than sorry.

Grabelkovsky, 2019-07-09:
Stephen,

1. Thanks a lot for replay.
2. It would be reasonable, if Microsoft add appropriate information in MSDN.

I guessed DPC, you propose even IRQ.
I plan to see inside source in all cases. Theoretically this information
may be calculated without any system switching inside, so IRQ is
possible... :)

Regards,
Michael.

stephenscalzo, 2019-07-09:
I agree with everything you've said.

I've attached an image of PsGetCurrentThreadTeb decompilation in IDA Pro. I slightly cleaned it up for you.

[image]

The version of ntoskrnl.exe it is for is Windows 10 Enterprise (1809).

1. Check the MiscFlags field of the _KTHREAD structure for the flags 0x400.
2. Check the ApcStateIndex field of the _KTHREAD structure.
3. If the flag was present in #1 or the ApcStateIndex field evaluates to true then return a null pointer.
4. Else, return the pointer to the _TEB structure for the current thread object.

If you hadn't seen this yet then it might help you.

I hope someone from Microsoft tunes into this ticket soon because I'm also interested in the response.

Grabelkovsky, 2019-07-10:
Stephen,
thanks a lot!

It help, thank You.
Because KeGetCurrentThread has IRQL "Any Level, means PsGetCurrentThreadTeb
too...:)

I don't understand only way sometimes it returns null, but it is other
question...

Regards & Thanks a lot,
Michael.

tedhudek, 2019-10-16:
Hi @Grabelkovsky, @stephenscalzo, apologies for the delay.

I just updated https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/nf-ntddk-psgetcurrentthreadteb with the guidance you requested.

Specifically, see the Return Value section, and the new Remarks section.

Please let me know if you still have any issues on this, and thanks again for the feedback.

closed 2019-10-16 by tedhudek