MicrosoftDocs/windows-driver-docs-ddi #590: All links to the HidP_ functions are broken. 404 not found. HidP_GetData, HidP_GetValueCaps, etc, etc.

tkktkt, 2019-07-17:
All links to the HidP_ functions are broken. 404 not found. HidP_GetData, HidP_GetValueCaps, etc, etc.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 120432f3-6bb5-4644-571e-ea8db1823953
* Version Independent ID: 839799df-edae-33a2-d637-ae2912551787
* Content: [Human Interface Devices (HID) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/_hid/#feedback)
* Content Source: [wdk-ddi-src/content/_hid/index.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/_hid/index.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @cdwilson
* Microsoft Alias: **windowsdriverdev**

duncanmacmichael, 2019-07-18:
Hi @tkktkt, sorry for the inconvenience here. This is a known doc bug where the entire Hidpi.h header isn't rendering, and I'm working on a solution internally to get that working right now. I'll be sure to let you know when we have the docs available.

duncanmacmichael, 2019-07-29:
Hi @tkktkt , thanks for your patience. We've resolved the internal bug that was preventing the Hidpi.h header's content from rendering, so all links are now live.

closed 2019-07-29 by duncanmacmichael