MicrosoftDocs/windows-driver-docs-ddi #592: This function might be called at APC_LEVEL due to a bug on win8/8.1

yoavwinstein, 2019-07-21:
Hi, i was able to produce s crash dump similliar to this one:
https://community.osr.com/discussion/248406/why-is-my-pssetcreatethreadnotifyroutine-callback-called-at-apc-level

Mine had an ob callback on the stack instead of a thread notify routine.

I was able to reproduce this bug by writing data to 2 volumes during shutdown on a win8 x64 machine with driver verifier turned on
Also, it seems like you've updated the notify routine's doc but not the ob callback's

Bottom line: i think there should be a doc update about this routine being called at APC_LEVEL on some very rare occasions on win8/8.1

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6785b9ca-5aa2-9dd7-0cbe-97b71ef20858
* Version Independent ID: 473199f0-25df-9b2c-8088-1e1c839ecd6c
* Content: [POB_PRE_OPERATION_CALLBACK (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nc-wdm-pob_pre_operation_callback#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nc-wdm-pob_pre_operation_callback.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nc-wdm-pob_pre_operation_callback.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-09-23:
Hi @yoavwinstein thanks for the feedback, and apologies for the delay getting back to you.  Can you give me more info on which very rare occasions you mean?  Better yet, open a pull request against the doc page and propose the exact change yourself :)

yoavwinstein, 2019-09-24:
hi ted, i'll open a pr proposing changes as soon as i have time(probably this weekend)
the rare occasion is when there's more than one volume which has flush pending, that should trigger creation of one or more worker threads flushing volumes to disk. The bug is that the call to `PsCreateSystemThread`, which should be done in `IRQL=PASSIVE_LEVEL` is done in `IRQL=APC_LEVEL` which in turn calls notify routines and `ObRegisterCallback` callbacks

tedhudek, 2019-10-18:
Thanks @yoavwinstein sorry, just circling back on this.  Also, in your initial post, you said `Also, it seems like you've updated the notify routine's doc` -- which page were you referring to?

tedhudek, 2019-10-28:
Hi @yoavwinstein, I'm going to go ahead and close this one out for now, since it seems to be a pretty edge case.  If you still feel the caveat is important to get on the doc page, feel free to submit a PR.  I'll get it reviewed internally before it gets published.  Thanks again for caring about the docs :)

closed 2019-10-28 by tedhudek