MicrosoftDocs/windows-driver-docs-ddi #593: Causes memory access error if the FO is not associated with a KS object

emuzychenko, 2019-07-21:
The function doen't check if the FO is associated with a KS object, and may cause memory
access error otherwise.

It makes a sense to at least mention such behavior in the documentation, because there is no legal way to check if the FO is associated with a KS object other than using SEH to catch the possible exception.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0b919546-d82f-a0fe-76aa-22be7e2203f0
* Version Independent ID: 13665a6b-0eb5-83e8-f716-47e0339eb3d0
* Content: [KsGetObjectTypeFromFileObject function (ks.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ks/nf-ks-ksgetobjecttypefromfileobject)
* Content Source: [wdk-ddi-src/content/ks/nf-ks-ksgetobjecttypefromfileobject.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ks/nf-ks-ksgetobjecttypefromfileobject.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2021-01-29:
@emuzychenko, thanks for pointing out that this function does not check to see if the file object is associated with a KS object, which may cause memory access errors.

I'm adding this feedback to an internal work item to correct and update the KsGetObjectTypeFromFileObject function topic.

barrygolden, 2021-04-12:
Merged PR 17536: 372065, 372066, 372067

Provided driver developer guidance regarding use of structured exception handling to catch memory access errors when a file object or IRP is not associated with a KS object.

Reference wdk-ddi repo GitHub Issues 593, 594, 595

closed 2021-04-12 by barrygolden