MicrosoftDocs/windows-driver-docs-ddi #594: Causes memory access error if the IRP is not associated with a KS object

emuzychenko, 2019-07-21:
The function doen't check if the IRP is associated with a KS object, and may cause memory
access error otherwise.

It makes a sense to at least mention such behavior in the documentation, because there is no legal way to check if the IRP is associated with a KS object other than using SEH to catch the possible exception.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1a687c6e-4dbb-b7b2-a016-9994f3e2938a
* Version Independent ID: b06e0ed7-f41a-4d83-c0c3-b28405df6eff
* Content: [KsGetObjectTypeFromIrp function (ks.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ks/nf-ks-ksgetobjecttypefromirp)
* Content Source: [wdk-ddi-src/content/ks/nf-ks-ksgetobjecttypefromirp.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ks/nf-ks-ksgetobjecttypefromirp.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2021-01-29:
@emuzychenko, thanks for pointing out that this function does not check to see if the IRP is associated with a KS object, which may cause memory access errors.

I'm adding this feedback to an internal work item to correct and update the KsGetObjectTypeFromIrp function topic.

barrygolden, 2021-04-12:
Merged PR 17536: 372065, 372066, 372067

Provided driver developer guidance regarding use of structured exception handling to catch memory access errors when a file object or IRP is not associated with a KS object.

Reference wdk-ddi repo GitHub Issues 593, 594, 595

closed 2021-04-12 by barrygolden