MicrosoftDocs/windows-driver-docs-ddi #597: Accessing registry

JonAxtell, 2019-07-23:
"print monitors must not explicitly call either the Win32 registry API" Why not? 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 46651cb5-11e0-4298-f3d1-5fe4d1359890
* Version Independent ID: deefa12a-e34f-f051-8403-2c28bf5c490f
* Content: [_MONITORREG (winsplp.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/winsplp/ns-winsplp-_monitorreg)
* Content Source: [wdk-ddi-src/content/winsplp/ns-winsplp-_monitorreg.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/winsplp/ns-winsplp-_monitorreg.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

hsebs, 2020-10-31:
@JonAxtell 
The spooler keeps a copy of the registry inside.
So, you shouldn't use the Win32 registry API to get the value actually used by the spooler.

barrygolden, 2020-11-04:
Thanks for the clarification, @hsebs!  I'll add a note to the topic that encompasses your explanation.

closed 2020-11-04 by barrygolden