MicrosoftDocs/windows-driver-docs-ddi #598: Members of structure

JonAxtell, 2019-07-23:
Some more detail, such as brief description link to page giving fuller description & usage required. Also details about how and when and why the spooler calls the functions and what it expects the UI to do.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b287bef8-87ec-19e4-b7aa-7aa5b9896198
* Version Independent ID: 0ceecce9-a675-c929-1025-2c20e9a562fd
* Content: [_MONITORUI (winsplp.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/winsplp/ns-winsplp-_monitorui)
* Content Source: [wdk-ddi-src/content/winsplp/ns-winsplp-_monitorui.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/winsplp/ns-winsplp-_monitorui.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2021-01-29:
@JonAxtell, thanks for your feedback and for pointing out the missing information.

I'm opening an internal work item to fill in missing member descriptions and update the MONITORUI structure topic.

barrygolden, 2021-03-29:
Updated topic: filled in missing member function pointer descriptions.

closed 2021-03-29 by barrygolden