MicrosoftDocs/windows-driver-docs-ddi #599: What's With the Quotes?

PeterGV, 2019-07-23:
This isn't specific JUST to this page... but I've recently notice quotation marks showing up around the IRQL specification in the Requirements section.  As in:

        IRQL          "&lt;= DISPATCH_LEVEL"

What *possible* purpose do these quotation marks serve, except to make the documentation harder to read and cause people to wonder? 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a7eca031-4926-2994-1ba6-1cdf8d4207ef
* Version Independent ID: 310d6432-bae2-9f39-d69e-d355837f49e9
* Content: [WdfRequestForwardToIoQueue function (wdfrequest.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdfrequest/nf-wdfrequest-wdfrequestforwardtoioqueue#feedback)
* Content Source: [wdk-ddi-src/content/wdfrequest/nf-wdfrequest-wdfrequestforwardtoioqueue.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfrequest/nf-wdfrequest-wdfrequestforwardtoioqueue.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-08-06:
Thanks for the report @PeterGV.  I think I've got all of these removed now.  They were artifacts of the doc migration from xml to markdown.

closed 2019-08-06 by tedhudek

PeterGV, 2019-08-06:
LOL... I *knew* it had  to be something like that.

Thanks Ted... Always a pleasure to interact with you!