MicrosoftDocs/windows-driver-docs-ddi #612: where is Timeout parameter?

kouzhudong, 2019-08-08:
If the caller is executing at IRQL = DISPATCH_LEVEL or in an arbitrary thread context, the Timeout parameter to KeWaitXxx must be zero.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3eca21d5-ff25-f02c-28a1-7308378ab609
* Version Independent ID: 83894e3c-728e-dc0c-9c23-09fa790731cb
* Content: [KeReleaseSemaphore function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-kereleasesemaphore#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-kereleasesemaphore.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-kereleasesemaphore.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

kouzhudong, 2019-08-08:
in KeWaitXxx.

kouzhudong, 2019-08-08:
这句话不应该出现在这里。应该出现在KeWaitXxx。出现在这里是友好的提示。

tedhudek, 2019-10-17:
Thanks @kouzhudong I just removed that sentence.  I also verified that the same point already exists on the doc pages for **KeWaitForMultipleObjects** and **KeWaitForSingleObject**.  The change should appear on the live site within half an hour.

Thanks for taking the time to report this issue to us.

closed 2019-10-17 by tedhudek