MicrosoftDocs/windows-driver-docs-ddi #616: link is 404

dgkim20, 2019-08-12:
https://docs.microsoft.com/windows/desktop/api/ntdef/ns-ntdef-_list_entry
check link plz

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d8fa9c5e-355f-16db-1948-f094ab67751c
* Version Independent ID: 1053b3ae-c27e-397c-992d-0168dbc9916d
* Content: [ExInterlockedRemoveHeadList function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-exinterlockedremoveheadlist#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-exinterlockedremoveheadlist.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-exinterlockedremoveheadlist.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-10-17:
Thanks for the heads-up @dgkim20 and apologies for the delay.

I found 42 occurrences of that bad link across the repo, and just pushed the fix to update them to https://docs.microsoft.com/windows/win32/api/ntdef/ns-ntdef-list_entry.

The fix should appear on the live site within half an hour.  Appreciate the feedback!

closed 2019-10-17 by tedhudek

dgkim20, 2019-10-17:
thanks for your support. 
I confirmed the link. 
thank you.