MicrosoftDocs/windows-driver-docs-ddi #627: Does this structure really have three union members with the same name

peterwie, 2019-09-03:
The Syntax block indicates that there are 3 members named V1 with the same type.  THat doesn't seem possible.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d0a1c23a-2a25-b9e7-87cf-d80a7e695507
* Version Independent ID: 1d405a87-ee4c-20f7-3f3f-b81b2c523445
* Content: [_DMA_ADAPTER_INFO (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/ns-wdm-_dma_adapter_info)
* Content Source: [wdk-ddi-src/content/wdm/ns-wdm-_dma_adapter_info.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_dma_adapter_info.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

KalleOlaviNiemitalo, 2020-05-26:
This seems to have been fixed. The syntax block at [_DMA_ADAPTER_INFO (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/ns-wdm-_dma_adapter_info) no longer indicates three members named `V1`:

```C
typedef struct _DMA_ADAPTER_INFO {
  ULONG Version;
  union {
    DMA_ADAPTER_INFO_V1 V1;
  };
} DMA_ADAPTER_INFO, *PDMA_ADAPTER_INFO;
```

tedhudek, 2020-05-26:
Thanks @KalleOlaviNiemitalo for spotting the fix, and thanks @peterwie for the feedback

closed 2020-05-26 by tedhudek