MicrosoftDocs/windows-driver-docs-ddi #631: Invalid Declaration of SMARTCARD_EXTENSION.ReaderFunction

dwkaiser, 2019-09-06:
Current Declaration is:

NTSTATUS(                 *ReaderFunction[16];

Declaration Should Read:

NTSTATUS (*ReaderFunction[16])(PSMARTCARD_EXTENSION);

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: cbd9809e-57d5-a8b2-e9ba-78f0e3f9643a
* Version Independent ID: 8c83c33c-a0ab-1f14-50eb-c938e56683c3
* Content: [_SMARTCARD_EXTENSION (smclib.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/smclib/ns-smclib-_smartcard_extension)
* Content Source: [wdk-ddi-src/content/smclib/ns-smclib-_smartcard_extension.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/smclib/ns-smclib-_smartcard_extension.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-09-06:
Working on the fix. Thanks, @dwkaiser

EliotSeattle, 2019-10-22:
Still waiting for the build team to fix the tool that generates the syntax block. In the meantime, I added the info manually to the member description.
Sorry for the delay.
Eliot

closed 2019-10-22 by EliotSeattle