MicrosoftDocs/windows-driver-docs-ddi #632: The SampleSize field meaning is unclear

emuzychenko, 2019-09-07:
For audio formats, the word "sample" can mean both an individual sample (amplitude value) and a group of such samples for all stream channels (a frame). The nSamplesPerSec in WAVEFORMATEX actually means "frames per sec", while wBitsPerSample means individual sample size.

Here SampleSize is mentioned as a frame size field:

https://docs.microsoft.com/en-us/windows-hardware/drivers/stream/output-streams

So the meaning of this field definitely needs a clarification.

Additionally, this article requires SampleSize to be nonzero for fixed sample size formats. But a series of other articles suggests to set it to 0 for PCM audio formats, for example:

https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/pcm-stream-data-format

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7b68b411-7ba3-7a5c-4706-0d6151fbb53a
* Version Independent ID: 3caab1de-54a0-9d15-0b2d-a280293228f0
* Content: [KSDATAFORMAT (ks.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ks/ns-ks-ksdataformat)
* Content Source: [wdk-ddi-src/content/ks/ns-ks-ksdataformat.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ks/ns-ks-ksdataformat.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2020-12-17:
Adding DOMARS as a reviewer on this issue as this appears to be audio related.

barrygolden, 2021-01-29:
@emuzychenko, thanks for pointing out the ambiguity around the SampleSize member description.

I'm adding the feedback you've provided to an internal work item to correct and update the KSDATAFORMAT structure topic.

closed 2021-01-29 by barrygolden