MicrosoftDocs/windows-driver-docs-ddi #637: What are the differences between this and the non-ex version?

peterwie, 2019-09-16:
The text says:

BuildScatterGatherListEx is an extended version of the BuildScatterGatherList routine. The following list summarizes the features that are available only in the extended version:

but then there is no accompanying list.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 83dacef7-5ebf-6ef4-7012-a47fdd2ed9b6
* Version Independent ID: da10c98e-06a5-39b6-8354-f51c323d6955
* Content: [PBUILD_SCATTER_GATHER_LIST_EX (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nc-wdm-pbuild_scatter_gather_list_ex)
* Content Source: [wdk-ddi-src/content/wdm/nc-wdm-pbuild_scatter_gather_list_ex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nc-wdm-pbuild_scatter_gather_list_ex.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

EliotSeattle, 2019-09-17:
Thanks for catching this migration issue, @peterwie 
The table is now in the live topic.
I have an opening for a good driver doc writer coming up. May be a step down in pay, but we don't lack in work. :-)
Again, thanks!

closed 2019-09-17 by EliotSeattle

peterwie, 2019-09-17:
Thanks Eliot.  I’m better at criticizing other people’s writing than doing my own 😊

-p