MicrosoftDocs/windows-driver-docs-ddi #648: Link points to the wrong page

shaboinkin, 2019-09-30:
"The ACPI notification ID. This parameter specifies the type of ACPI notification that PoFx is sending. For a list of ACPI notification IDs, see ACPI notifications."

The link "ACPI notifications" points to the page "Programming reference for Windows Driver Kit ". I would have expected it to show me stuff about ACPI notifications. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a8a885ca-e592-d974-4651-07b51a6c1ce1
* Version Independent ID: dd70ab29-cf5e-51d7-a6d5-5889d5ba81b8
* Content: [PEPCALLBACKNOTIFYACPI (pepfx.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/pepfx/nc-pepfx-pepcallbacknotifyacpi#feedback)
* Content Source: [wdk-ddi-src/content/pepfx/nc-pepfx-pepcallbacknotifyacpi.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/pepfx/nc-pepfx-pepcallbacknotifyacpi.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

shaboinkin, 2019-09-30:
The "power management framework" link also points the the WDK docs page.

tedhudek, 2019-10-17:
Thanks @shaboinkin for the report.  I just updated the link to point to https://docs.microsoft.com/windows-hardware/drivers/kernel/acpi-notifications.

I also added a link from https://docs.microsoft.com/windows-hardware/drivers/kernel/acpi-notifications back to https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/pepfx/nc-pepfx-pepcallbacknotifyacpi.

Appreciate the feedback!

closed 2019-10-17 by tedhudek