MicrosoftDocs/windows-driver-docs-ddi #652: What should the type of CurrentCount be?

rahul-ramadas, 2019-10-04:
Type of CurrentCount is unspecified.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4e1f35e7-8f90-3b85-6299-d4b6cb9a23bf
* Version Independent ID: 9920e204-b4a4-c591-b5e9-f1e77391f1e5
* Content: [KeQueryTickCount macro (ntddk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/nf-ntddk-kequerytickcount?redirectedfrom=MSDN)
* Content Source: [wdk-ddi-src/content/ntddk/nf-ntddk-kequerytickcount.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nf-ntddk-kequerytickcount.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

DoaJCBlogger, 2019-10-17:
According to the Windows 7 WDK documentation, TickCount is defined as "OUT PLARGE_INTEGER TickCount".

tedhudek, 2019-10-28:
Fixed.  Thanks @rahul-ramadas and @DoaJCBlogger

closed 2019-10-28 by tedhudek