MicrosoftDocs/windows-driver-docs-ddi #657: FILE_INFORMATION_CLASS定义的问题。

kouzhudong, 2019-10-12:
[Enter feedback here]

FILE_INFORMATION_CLASS是不是加个版本控制的宏定义呢？

有好些定义，特别是后面的，在以前的WDK的头文件都没有定义和以前的操作操作系统都没实现。

FileIoCompletionNotificationInformation | This value is reserved for system use. This value is available starting with Windows Vista.
这些说明最好在代码（头文件的定义）中体现，你们说是不是？

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b2d450a1-f049-5a37-fe23-76b369d89738
* Version Independent ID: 2b528307-4778-6bd7-a10c-d6d518f7df98
* Content: [_FILE_INFORMATION_CLASS (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/ne-wdm-_file_information_class)
* Content Source: [wdk-ddi-src/content/wdm/ne-wdm-_file_information_class.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ne-wdm-_file_information_class.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-10-14:
English translation:

FILE_INFORMATION_CLASS
Is it a macro definition with version control? There are a number of definitions, especially later, in the previous WDK header file is not defined and the previous operating system is not implemented.

FileIoCompletionNotificationInformation | This value is reserved for system use. This value is available starting with Windows Vista.

These instructions are best reflected in the code (the definition of the header file), do you say?

tedhudek, 2019-10-14:
Hi @kouzhudong, sorry, I used Bing to translate to English but I'm not sure what you are asking.  Can you be specific with your request?  Thanks

tedhudek, 2019-10-16:
Closing this one out for now.  Please reopen with more specifics as needed.

closed 2019-10-16 by tedhudek